joshnishikawa/MIDIcontroller #4: MIDI Pot smooth does not work at all.

johannthorir, 2020-01-11:
See comments in code:

int MIDIpot::smooth(int val, int NR){
  buffer = 0; // set to zero so any memory gone.
  balancedValue = 0; // this is set to 0 here so any memory is gone
  difference = val - balancedValue;  //  now difference is equal to val - 0 which is val
  // if val is 0 then buffer = buffer/2, which is 0 which is val
  // if val is not 0  then buffer = buffer + difference which is 0 + val which is val.
  // from this you can see that in all cases buffer == val
  buffer = val == balancedValue ? buffer/2 : buffer+difference;

  // if the square of the buffer which is the square of val is greater than the square of NR
  // then you return val 
  //  you return 0 however for any value of val which is less than the sqrt of NR.
  
  if (buffer*buffer > NR*NR){ // This works better than abs(buffer) for me.
    balancedValue = val;
    buffer = 0;
  }
  return balancedValue;
};

the only use of the smooth member function is like this:
this->smooth(analogRead(pin), 100);
so smooth does nothing except clamp all values less than 100 to zero with no actual smoothing.

joshnishikawa, 2020-01-11:
Yes the variables are initialized to zero but they're updated every time the function is called. Have you actually tested this with an analog input and it's not working? Or are you just looking through the code?  In the examples folder, there is an image called "AnalogToMidi.jpg" which shows actual serial data recorded during testing. If you're not getting similar test results, please provide more details about your setup and exactly what results you *are* getting.

rustybrooks, 2020-01-11:
I haven't tried it either, but look at the code - all the class variables are initialized to constants every time smooth() gets called.  I don't see how it could possibly work.

rustybrooks, 2020-01-11:
(I think the function will more or less "work" in that it will return values if they're larger than NR, but they won't be smoothed)

joshnishikawa, 2020-01-11:
Ah! I see. They're also being initialized **in** the function as if they were static variables (that's what they were originally). I didn't remove those lines when I made them class variables. Oops! Thank you for pointing this out. I was able to fix a couple of other things too.

rustybrooks, 2020-01-11:
Yeah that makes sense, I kind of thought it might be something like that.

closed 2020-01-12 by joshnishikawa