joshnishikawa/MIDIcontroller #5: 'elapsedMillis' does not name a type Error

MaliceDee, 2020-01-17:
Greetings, 
Have been trying to use this library but haven't able to get any of the examples to compile, I keep getting this error; 

``Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino Leonardo"

In file included from C:\Users\user\Documents\Arduino\libraries\Flicker-master/Flicker.h:41:0,

                 from C:\Users\user\Documents\Arduino\libraries\MIDIcontroller-master/MIDIbutton.h:6,

                 from C:\Users\user\Documents\Arduino\libraries\MIDIcontroller-master/MIDIcontroller.h:5,

                 from C:\Users\user\Documents\Arduino\libraries\MIDIcontroller-master\examples\afterTouch\afterTouch.ino:1:

C:\Users\user\Documents\Arduino\libraries\Flicker-master/TouchSwitch.h:63:5: error: 'elapsedMillis' does not name a type

     elapsedMillis held_millis = 0;

     ^~~~~~~~~~~~~

In file included from C:\Users\user\Documents\Arduino\libraries\Flicker-master/Flicker.h:43:0,

                 from C:\Users\user\Documents\Arduino\libraries\MIDIcontroller-master/MIDIbutton.h:6,

                 from C:\Users\user\Documents\Arduino\libraries\MIDIcontroller-master/MIDIcontroller.h:5,

                 from C:\Users\user\Documents\Arduino\libraries\MIDIcontroller-master\examples\afterTouch\afterTouch.ino:1:

C:\Users\user\Documents\Arduino\libraries\Flicker-master/TouchVelocity.h:66:5: error: 'elapsedMicros' does not name a type

     elapsedMicros timer;

     ^~~~~~~~~~~~~

In file included from C:\Users\user\Documents\Arduino\libraries\MIDIcontroller-master/MIDIcontroller.h:9:0,

                 from C:\Users\user\Documents\Arduino\libraries\MIDIcontroller-master\examples\afterTouch\afterTouch.ino:1:

C:\Users\user\Documents\Arduino\libraries\MIDIcontroller-master/MIDIdrum.h:20:5: error: 'elapsedMillis' does not name a type

     elapsedMillis timer;

     ^~~~~~~~~~~~~``

I'm not sure how to fix it, I have all the appropriate libraries installed? 
Any help would be immensely appreciated, this library looks like a lot of fun to play with if  I can get it working :)

 Regards
-Jim Foster

joshnishikawa, 2020-01-17:
This library was written specifically for the "Teensy" line of microcontrollers and those are the only thing I've personally tested on. I have no idea how well the library works on Leonardo (if at all) but I hope you do get it working. I'm sure elapsedMillis and elapsedMicros is a part of the Teensy core library but you may need to install the library manually for others. It's here https://github.com/pfeerick/elapsedMillis/archive/master.zip

MaliceDee, 2020-01-18:
I have been intending modify the sketches / .h files to make a version of the library that would send Midi through an arduino friendly library so i could implement these tools easily in my arduino sketches, I just hoped to be able to do this without having to change too much else, I'm a bit fresh into learning all this so its hard to work out where I've gone wrong when i tinker too much with these things but it looks like I'll be rewriting the elapsedMillis/ Micros functions with the arduino friendly equivalents  and hopefully I can get it working 👍  Thank you for your time, very much appreciated!

closed 2020-01-19 by MaliceDee