joshnishikawa/MIDIcontroller #6: Teensy 4 and Touch

Troub99, 2020-03-09:
Hi Josh
I'm building a midi controller with encoders and buttons. I'm attempting to use a Teensy 4 along with your library. I can compile the sketch in Arduino IDE for Pots and Encoders, but the button example gives me this error with the Teensy 4 (it compiles fine for Teensy 3.6):

> Arduino: 1.8.11 (Mac OS X), TD: 1.51, Board: "Teensy 4.0, MIDI, 600 MHz, Faster, US English

/var/folders/j0/v7q8w86j5lqfx_ph7blxdf600000gn/T/arduino_build_188966/libraries/Flicker-master/TouchSwitch.cpp.o: In function `TouchSwitch::trigger()':
Documents/Arduino/libraries/Flicker-master/TouchSwitch.cpp:88: undefined reference to `touchRead'
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 4.0.

I've been trying to take the touch functions out of the library and button files as I don't need them for this project. I feel I've been getting close, but no success so far. Can you help at all with how I can modify the button.h and button.cpp files? Thank you.

joshnishikawa, 2020-03-10:
Thank you! The MIDIbutton class supports capacitive touch but Teensy 4.0 doesn't. I'll have to throw some 'if' statements in there. Stand by.

joshnishikawa, 2020-03-10:
I put a band-aid on the MIDIbutton and MIDIdrum read functions so that anything that references touchRead() is only compiled if the board is not a 4.0. I'll leave this issue open until I have an opportunity to fix it properly but, for now, it should compile and run just fine for Teensy 4.0 users. Please let me know if it doesn't as I can only compile the code. I don't actually own a 4.0 to test myself :^(

Troub99, 2020-03-10:
That's fantastic, thank you. I'll give it a go and report back.

Troub99, 2020-03-11:
Working fantastically with Teensy 4 now, thank you. I expanded your MIDIEnc example to 8 rotary encoders with switches and my controller is now flying. I am extremely grateful.

closed 2023-03-22 by joshnishikawa