joshnishikawa/MIDIcontroller #8: easy way to incorporate this library with a button matrix?

stonepreston, 2020-09-23:
This library looks great! Glossing over the examples and it seems like the button objects are pretty much configured to be directly connected to a pin, is there a way to get things working with a matrix of buttons?

joshnishikawa, 2020-09-24:
There's not currently an easy way to setup a matrix of buttons. It seems like there should be but there are a few different ways to wire up a matrix depending on what hardware you're using or whether multiple buttons on the same row/column should work at the same time etc. I've never actually made a button matrix but I'll try it when I get a chance. Hopefully I can come up with something simple enough to include in this library. Any info or insight is appreciated.

stonepreston, 2020-09-24:
That would be awesome. I feel like any midi controller thats more than a few knobs and buttons would be using a matrix since it saves so many pins. I have not actually made one yet either haha but Ive done lots of research. For matrix scanning there is this [keypad library](https://github.com/nickgammon/Keypad_Matrix) that seems very easy to use to get a matrix up and running. 

For general theory about button matrices the mechanical keyboard community has a wealth of resources. [Here (https://beta.docs.qmk.fm/developing-qmk/for-a-deeper-understanding/how_a_matrix_works) is a good overview of the matrix, and [here](https://beta.docs.qmk.fm/using-qmk/guides/keyboard-building/hand_wire) is a nice guide on handwiring and different ways of soldering the diodes and what not.

joshnishikawa, 2021-03-30:
It looks like the 'Keypad' library has button matrixes pretty well sorted out. It should be just a matter of writing a sketch that uses the Keypad library to send MIDI messages instead of key presses. It would be interesting to see a proof of concept. However, increasing the number of MIDI inputs is generally done with MUX chips as they are very cheap and simple to implement. I've added a sketch for muxed inputs to the examples.

closed 2021-03-30 by joshnishikawa