joshnishikawa/MIDIcontroller #9: MIDIpot doesn't work with Teensy3.2

Phipanjo, 2021-02-24:
At compiling the IDE fails to run the program.
Would be too good to be true. The library would save me a lot of work.

I attach the error statement here:

`Arduino: 1.8.13 (Mac OS X), TD: 1.53, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz (overclock), Faster, US English"

/Applications/Teensyduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Teensyduino.app/Contents/Java/hardware -hardware /Users/philiphaas/Library/Arduino15/packages -tools /Applications/Teensyduino.app/Contents/Java/tools-builder -tools /Applications/Teensyduino.app/Contents/Java/hardware/tools/avr -tools /Users/philiphaas/Library/Arduino15/packages -built-in-libraries /Applications/Teensyduino.app/Contents/Java/libraries -libraries /Users/philiphaas/Documents/PCB/Arduino/libraries -fqbn=teensy:avr:teensy31:usb=serial,speed=96,opt=o2std,keys=en-us -ide-version=10813 -build-path /var/folders/9x/73_mykmd6l9ftyg780_8g4k40000gn/T/arduino_build_382262 -warnings=none -build-cache /var/folders/9x/73_mykmd6l9ftyg780_8g4k40000gn/T/arduino_cache_115385 -verbose /Users/philiphaas/Documents/PCB/Arduino/Projekte/Teensy/MIDIpot/MIDIpot.ino
/Applications/Teensyduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Teensyduino.app/Contents/Java/hardware -hardware /Users/philiphaas/Library/Arduino15/packages -tools /Applications/Teensyduino.app/Contents/Java/tools-builder -tools /Applications/Teensyduino.app/Contents/Java/hardware/tools/avr -tools /Users/philiphaas/Library/Arduino15/packages -built-in-libraries /Applications/Teensyduino.app/Contents/Java/libraries -libraries /Users/philiphaas/Documents/PCB/Arduino/libraries -fqbn=teensy:avr:teensy31:usb=serial,speed=96,opt=o2std,keys=en-us -ide-version=10813 -build-path /var/folders/9x/73_mykmd6l9ftyg780_8g4k40000gn/T/arduino_build_382262 -warnings=none -build-cache /var/folders/9x/73_mykmd6l9ftyg780_8g4k40000gn/T/arduino_cache_115385 -verbose /Users/philiphaas/Documents/PCB/Arduino/Projekte/Teensy/MIDIpot/MIDIpot.ino
Using board 'teensy31' from platform in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr
Using core 'teensy3' from platform in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr
Detecting libraries used...
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=153 -DARDUINO=10813 -DARDUINO_TEENSY32 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 /var/folders/9x/73_mykmd6l9ftyg780_8g4k40000gn/T/arduino_build_382262/sketch/MIDIpot.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for MIDIcontroller.h: [MIDIcontroller]
ResolveLibrary(MIDIcontroller.h)
  -> candidates: [MIDIcontroller]
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=153 -DARDUINO=10813 -DARDUINO_TEENSY32 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 -I/Users/philiphaas/Documents/PCB/Arduino/libraries/MIDIcontroller /var/folders/9x/73_mykmd6l9ftyg780_8g4k40000gn/T/arduino_build_382262/sketch/MIDIpot.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Bounce.h: [Bounce]
ResolveLibrary(Bounce.h)
  -> candidates: [Bounce]
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=153 -DARDUINO=10813 -DARDUINO_TEENSY32 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 -I/Users/philiphaas/Documents/PCB/Arduino/libraries/MIDIcontroller -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Bounce /var/folders/9x/73_mykmd6l9ftyg780_8g4k40000gn/T/arduino_build_382262/sketch/MIDIpot.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Flicker.h: []
ResolveLibrary(Flicker.h)
In file included from /Users/philiphaas/Documents/PCB/Arduino/libraries/MIDIcontroller/MIDIcontroller.h:5:0,  -> candidates: []

                 from /Users/philiphaas/Documents/PCB/Arduino/Projekte/Teensy/MIDIpot/MIDIpot.ino:1:
/Users/philiphaas/Documents/PCB/Arduino/libraries/MIDIcontroller/MIDIbutton.h:6:21: fatal error: Flicker.h: No such file or directory
compilation terminated.
Bibliothek MIDIcontroller im Ordner: /Users/philiphaas/Documents/PCB/Arduino/libraries/MIDIcontroller (legacy) wird verwendet
Bibliothek Bounce im Ordner: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Bounce (legacy) wird verwendet
Fehler beim Kompilieren für das Board Teensy 3.2 / 3.1.
`

joshnishikawa, 2021-02-24:
It looks like maybe you don't have the "Flicker" library installed. Pots don't really use it but the MIDIcontroller library needs it for cap touch stuff and complains if it's not there. I think there's a way I can set the MIDIcontroller library up to automatically install Flicker as a dependency. Until then you can install it from the Arduino Library Manager or [here](https://github.com/joshnishikawa/Flicker).

Phipanjo, 2021-02-24:
> It looks like maybe you don't have the "Flicker" library installed. Pots don't really use it but the MIDIcontroller library needs it for cap touch stuff and complains if it's not there. I think [...]

Genius!
Thank you josh, you safed my day! :-D

closed 2021-02-24 by Phipanjo