joshnishikawa/MIDIcontroller #10: Using a 4051 Multiplexer??

nimans4, 2021-03-17:
Is it possible to compliment your library to the usage of a 4051 chip?

For example the potentiometers:

You use <<MIDIpot myPot(potPin, 22);>> . But when I am using a 4051, 8 analogInputs will come via just one analog Input, due to the 4051. How will my PC Programm (Im working on a DJ Controller) distinguish between my 16 Potentiometers, so that every single potentiometer gets his own "mark"(?). Now, all Potentiometers connected to one 4051, will control all the same parameter in my PC Programm, right?

joshnishikawa, 2021-03-17:
Yeah, it'd definitely be worth throwing a simple mux example in there. I'll do that in the next update. Something like...
`setup(){
  pinMode(selector_pin_a, OUTPUT);
  pinMode(selector_pin_b, OUTPUT);
  pinMode(selector_pin_c, OUTPUT);

  for(int i=0; i<8; i++){
    Pots[i] = new MIDIpot(mux_input_pin, 20+i);
    // 20 starts a set of 8 CC numbers that are undefined in the MIDI spec.
    // Obviously you can use any CC number you like. A 'for' loop is just easy.
  }
}

loop(){
  for(int i=0; i<8; i++){
    digitalWrite(selector_pin_a, (i&7)>>2);
    digitalWrite(selector_pin_b, (i&3)>>1);
    digitalWrite(selector_pin_c, (i&1));

    Pots[i]->send();
  }  
}`

joshnishikawa, 2021-03-20:
The previous comment has been updated with a working example. That example has also been added to the library.

closed 2021-03-20 by joshnishikawa