joshnishikawa/MIDIcontroller #14: Use usbHost on Teensy 4.1?

Miq1, 2021-10-04:
This looks really promising for my application (I am building a multi-pad controller from scratch)!
I was wondering if and how it could be possible to use the USB host port on the Teensy 4.1 with the MIDIcontroller lib? I would like to keep the base USB port for debugging etc.

joshnishikawa, 2021-10-05:
You can definitely use the USB host along with the MIDIcontroller lib. It's just a matter of what you want to do. The 4.x uses the same USB host as the 3.6 so you can just start with that example. Here it is on github. [https://github.com/PaulStoffregen/USBHost_t36/blob/master/examples/MIDI/Basic/Basic.ino](url). The callback functions in this example just print info to the serial monitor but you could have those callbacks do anything you like. For example, I had a USB MIDI controller plugged into the USB host of a Teensy 3.6. That Teensy had other sensors hooked up to it that I could use to apply MIDI effects (i.e. octave, arpeggiator, delay etc.) and then pass the resulting MIDI messages on to my DAW. So both controllers only took up one USB port on my computer yet the signals were kept separate and could each be routed to multiple places. A simple (untested) octave callback would be...
 
```
void OnNoteOn(byte channel, byte note, byte velocity){
  // note the order of arguments differs between receiving and sending
  usbMIDI.sendNoteOn(note, velocity, channel);

  // send a second note one octave lower
  if (note >= 12){
      usbMIDI.sendNoteOn(note - 12, velocity, channel);
  }
}

```
Or you might not even plug the Teensy into a computer; Just have a professional USB MIDI controller hooked up to the Teensy host which applies any MIDI effects and then uses the Audio library to apply audio effects and output the audio directly to an amp. Like I say, it just depends on what you want to do.

Miq1, 2021-10-05:
Thanks a lot! I was erroneously assuming I had to use the USB host as a separate connection for the teensy to act as a MIDI controller, but the MIDI data will be sent as a different protocol on the regular Teensy USB port of course. So that opens up the USB host port for a usage like you described - great!

closed 2021-10-14 by joshnishikawa