joshnishikawa/MIDIcontroller #16: Encoder buttons

JukkaPVK, 2021-11-19:
Hi Josh,

I'm building a MIDI controller for personal use and found your library. It works nicely, thank you for your effort! I was wondering if it would be possible to program an encoder to send a keypress when a know (encoder) is turned? It could be handy for example programming scrolling or selecting channel in DAW, if needed.

joshnishikawa, 2021-11-22:
I know the issue is closed (you probably figured it out already) but I did want to put a response here for anyone else trying to do the same. That particular functionality could be implemented simply enough without this library. Please try the following...
`#include <Encoder.h>

Encoder myEnc(25, 24); // or whatever pins your encoder is on
byte MIDIchannel = 3;  // or whatever channel you like

void setup(){}

void loop(){
  int knob = myEnc.read();

  if (knob >= 4){
    usbMIDI.sendControlChange(14, 127, MIDIchannel);
    myEnc.write(0);
  }
  else if (knob <= -4 ){
    usbMIDI.sendControlChange(15, 127, MIDIchannel);
    myEnc.write(0);
  }
}`
You should then be able to assign a right turn to the 'next' button and a left turn to the 'prev' button...or any other assignment you wish to make. I could add an option to the MIDIenc class but it would break some functionality already in place and, as I said, it's simple enough to implement. I hope that helps.

closed 2021-11-22 by joshnishikawa

JukkaPVK, 2021-11-22:
Thanks for your reply. I actually made a fork of  this as I couldn't get the encoder to work with encoder by detent  (4 pulses). I made some modifications on myEnc.cpp to get it working. (I used pre-soldered Arduino encoder hw, when testing). Feel free comment / use the code if needed.

Also made myJog class for creating a simple jogwheel - Basically the same as encoder, but output a bit diffrent (not in GitHub yet). I also thought if it would be possible to create virtual encoders, i.e 1 encoder for e.g. 8 channels, and one for selecting channel but that's just an idea for small devices.

joshnishikawa, 2021-11-23:
Ah! Very nice catch. The call to `myKnob->write(0);` needed to be _inside_ the conditional statement of the `read()` function so that it actually allows the encoder value to reach 4 or -4. (See lines 53 and 57 of MIDIenc.cpp). It should be working now.

Encoders are so handy. I have one with a push-switch. The knob sends MIDI normally when turned. If I hold down the knob while turning, it changes the MIDI channel. If I press and release (without changing the MIDI channel) the knob toggles between completely different functionalities.

The jogwheel idea sounds very interesting. Good luck!

JukkaPVK, 2021-11-24:
Hi Josh,

Commenting to this old issue. Uploaded my version of jogwheel to my fork. Feel free add it to your library if you find it useful. Basically a copy of your good encoder library with a few tweaks.

Thanks,
Jukka