joshnishikawa/MIDIcontroller #19: enable to have a initialestate opf MIDIPot

mrbbp, 2022-04-23:
Hello sorry for this certainly basic question.

I try to send the initial value of MIDIPot (without any manipulation) when teensy received a noteOn event

```
void setup() {
  usbMIDI.setHandleNoteOn(OnNoteOn);
}

void OnNoteOn(byte channel, byte note, byte velocity) {
  digitalWrite(13, HIGH); // Any Note-On turns on LED
  for (int i = 0;  i < ANALOG_PINS_NBR; i++) {
    Pots[i]->send(); // notice the -> operator  
    delay(50);
  }
  digitalWrite(13, LOW); // Any Note-On turns on LED
}
```
The led HIG and LOW, but i do not received any value from midi.
What i'm doing wrong.
Do i have to read value with `analogRead(APin)`or have you a trick?

Oh and sorry to forget... Great job! thanks

regards

mrbbp, 2022-04-23:
I tried with `read()` and `send()`, but no more return.
if i turn the pot i received value

joshnishikawa, 2022-04-23:
I think I understand part of what you want to do...You want to bulk send the values of the pots to the DAW? The library doesn't currently handle that but it needs to. Bulk send really needs to be implemented. Give me a couple of days and I'll push an update.

It's interesting that you want to do the bulk send on Note ON...

mrbbp, 2022-04-23:
I do not use it with synth. ;)
I use midi controller with Processing.org code, to generate graphics...
Control or noteOn or onteOff, is the same for me. it's just an event. :)
I use [themidibus lib](https://github.com/sparks/themidibus) to play with midi in Processing.org

I will be wise and I will wait for the update

joshnishikawa, 2022-04-25:
Okay, I just pushed version 2.5.3 which includes an option to pass a boolean to the send function. Now you should be able to use `Pots[i]->send(FORCE)` to have a controller send its current value programmatically. This will work for MIDIpot, MIDIenc, MIDItouch and MIDIswitch (AKA MIDIbutton) classes. Please let me know if it works for you.

mrbbp, 2022-04-25:
Thank you very much.

i love the way you do this.
It works like a charm!

closed 2022-04-25 by joshnishikawa