joshnishikawa/MIDIcontroller #22: Arduino IDE 2.0.0 install issue

Marsupio, 2022-09-21:
I have just tried installing MIDIcontroller 2.5.3 but IDE just does nothing

joshnishikawa, 2022-09-21:
Thanks for the heads up. Yeah, 2.0 doesn't like the latest few versions for some reason. I posted the issue here if you want to keep an eye on it.
https://github.com/arduino/library-registry/issues/1895

Meanwhile, 1.8.19 still installs it fine. Or you can grab the [zip](https://github.com/joshnishikawa/MIDIcontroller/archive/refs/tags/2.5.3.zip) or the [tar.gz](https://github.com/joshnishikawa/MIDIcontroller/archive/refs/tags/2.5.3.tar.gz) and do it manually.

joshnishikawa, 2022-09-24:
This library depended on the Bounce library which is now deprecated in favor of Bounce2. Bounce has now been removed completely from the Library Manager and, when it tried to locate all MIDIcontroller's dependencies, it was failing (silently) to do so.

The suggested fix to remove the Bounce dependency entirely should be fine because most people probably already have Bounce2 installed. However, I included Bounce2 as a dependency because...well, this library depends on it.

Anyway, my understanding is that the Library Manager ignores changes unless you put out a new release. So, I did that (2.5.4) but I didn't want to close the issue until I was sure it worked. It does!

Please try installing version 2.5.4 from the Arduino IDE2 Library Manager.

closed 2022-09-24 by joshnishikawa