joshnishikawa/MIDIcontroller #26: Feature Request (Seeeduino Xiao)

digitalelements, 2022-11-26:
Hi Josh,

Is there any chance that the MIDIcontroller library might be able to be compatible with a board such as the Seeeduino Xiao ?
I have attempted using some of my previous sketches (for Teensy) and, MIDIdrum_FSR examples but I get a compilation error which references the old bounce.h. This board is much more readily available than say Teensy and much more cost effective for some of my super small drum projects. 

Regards,

Chris 

`In file included from /Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIcontroller.h:5:0,
                 from /private/var/folders/15/sz8y5tds6jg4lswv01_rv99w0000gn/T/.arduinoIDE-unsaved20221026-15329-j1uvv6.6hidc/MIDIdrum_FSR/MIDIdrum_FSR.ino:1:
/Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIswitch.h:5:10: fatal error: Bounce.h: No such file or directory
 #include "Bounce.h"
          ^~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1`

joshnishikawa, 2022-12-13:
Yes, I'm sure a lot of what's in this library can be used with other boards but can't be sure exactly what or how well because I only have Teensys to test with.

This particular error appears because the Bounce library is deprecated and older versions of the MIDIcontroller library still require it. Please update to version 2.5.4 (or higher) of the MIDIcontroller library. If you do so from the Arduino Library Manager, Bounce2 will be installed automatically. Otherwise you may need to install the Bounce2 library manually.

digitalelements, 2022-12-13:
Hi Josh, 

I'd be more than happy to facilitate getting a Seeeduino Xiao for you to test with if you were up for it ?

The funny thing is it appears my libraries are up to date. the compilation error only happens when I try to send my sketch to the Xiao board. I've even manually installed MIDIcontroller 2.5.5 as it does not show in the IDE 2 updates for MIDIcontroller. I've also checked and double checked that Bounce2 was installed a few times.

joshnishikawa, 2022-12-13:
Ah yes. I updated the dependency to Bounce2 but didn't actually update the line that includes it. I'll fix that bug and one other that I'm working on before I create the 2.5.5 release. Once that release is created, it will appear in the Arduino IDE too.

joshnishikawa, 2022-12-13:
For now you can just change line 5 of MIDIswitch.h to `#include "Bounce2.h"`

digitalelements, 2022-12-13:
Ok ... I did change the line but it appears it's a bit of a rabbit hole. the error has moved down to another .h file.
should I just follow the error and hunt for the includes ?

In file included from /Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIcontroller.h:7:0,
                 from /Users/chrisryan/Documents/Arduino/Xiao_FSR_Single/Xiao_FSR_Single.ino:1:
/Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIenc.h:5:10: fatal error: Bounce.h: No such file or directory
 #include "Bounce.h"
          ^~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

digitalelements, 2022-12-13:
I went ahead and updated the mentioned line for MIDIenc.h as well.

WARNING: library MIDIcontroller claims to run on avr architecture(s) and may be incompatible with your current board which runs on samd architecture(s).
WARNING: library Flicker claims to run on avr architecture(s) and may be incompatible with your current board which runs on samd architecture(s).
In file included from /Users/chrisryan/Documents/Arduino/libraries/Flicker/src/Flicker.h:5:0,
                 from /Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIswitch.h:6,
                 from /Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIcontroller.h:5,
                 from /Users/chrisryan/Documents/Arduino/Xiao_FSR_Single/Xiao_FSR_Single.ino:1:
/Users/chrisryan/Documents/Arduino/libraries/Flicker/src/TouchSwitch.h:63:5: error: 'elapsedMillis' does not name a type
     elapsedMillis held_millis = 0;
     ^~~~~~~~~~~~~
In file included from /Users/chrisryan/Documents/Arduino/libraries/Flicker/src/Flicker.h:7:0,
                 from /Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIswitch.h:6,
                 from /Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIcontroller.h:5,
                 from /Users/chrisryan/Documents/Arduino/Xiao_FSR_Single/Xiao_FSR_Single.ino:1:
/Users/chrisryan/Documents/Arduino/libraries/Flicker/src/TouchVelocity.h:67:5: error: 'elapsedMicros' does not name a type
     elapsedMicros timer;
     ^~~~~~~~~~~~~
In file included from /Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIcontroller.h:9:0,
                 from /Users/chrisryan/Documents/Arduino/Xiao_FSR_Single/Xiao_FSR_Single.ino:1:
/Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIdrum.h:20:5: error: 'elapsedMillis' does not name a type
     elapsedMillis timer;
     ^~~~~~~~~~~~~
/Users/chrisryan/Documents/Arduino/Xiao_FSR_Single/Xiao_FSR_Single.ino: In function 'void loop()':
/Users/chrisryan/Documents/Arduino/Xiao_FSR_Single/Xiao_FSR_Single.ino:46:9: error: 'usbMIDI' was not declared in this scope
   while(usbMIDI.read()){}
         ^~~~~~~

exit status 1

Compilation error: 'usbMIDI' was not declared in this scope

joshnishikawa, 2022-12-14:
> I went ahead and updated the mentioned line for MIDIenc.h as well.

Well done, I missed that one too. I went ahead and published the 2.5.5 release because this broken dependency really bugged me (pun intended).

However, looking at this list of errors makes it pretty clear that this library probably isn't going to work on a non-Teensy board. The usbMIDI library is part of the Teensy AVR core and the MIDIcontroller library depends entirely on it. In fact, the reason I originally wrote this library specifically for Teensy boards is because all that underlying code had already been written and was obviously well supported but only available on Teensy (not all Arduino compatible boards).

That is the code that would need to be written for a different chip architecture. I don't know if the XIAO core already has a MIDI library but, if not, that would definitely be beyond my current coding ability.

closed 2022-12-15 by joshnishikawa