joshnishikawa/MIDIcontroller #29: 3.1.0 Sensitivity

digitalelements, 2023-03-23:
I've starting working with my multi drum sketch again with Version 3.1.0. When I uploaded the new Library, I got the warning below which looked like i was trying to access a parameter that was private ? ( sensitivity ) 

``In file included from /Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIcontroller.h:9:0,
                 from /Users/chrisryan/Documents/Arduino/de_FSR_USB_Module_Sensi/de_FSR_USB_Module_Sensi.ino:1:
/Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIdrum.h: In function 'void setup()':
/Users/chrisryan/Documents/Arduino/libraries/MIDIcontroller/src/MIDIdrum.h:12:9: error: 'int MIDIdrum::sensitivity' is private
     int sensitivity;
         ^
/Users/chrisryan/Documents/Arduino/de_FSR_USB_Module_Sensi/de_FSR_USB_Module_Sensi.ino:39:7: error: within this context
  Pad0.sensitivity(80);
       ^
/Users/chrisryan/Documents/Arduino/de_FSR_USB_Module_Sensi/de_FSR_USB_Module_Sensi.ino:39:21: error: expression cannot be used as a function
  Pad0.sensitivity(80);
                     ^
Multiple libraries were found for "Bounce2.h"
  Used: /Users/chrisryan/Documents/Arduino/libraries/Bounce2
  Not used: /Users/chrisryan/Library/Arduino15/packages/teensy/hardware/avr/1.57.2/libraries/Bounce2
Multiple libraries were found for "Encoder.h"
  Used: /Users/chrisryan/Documents/Arduino/libraries/Encoder
  Not used: /Users/chrisryan/Library/Arduino15/packages/teensy/hardware/avr/1.57.2/libraries/Encoder
exit status 1

Compilation error: within this context``

I've also included my sketch for you review ... It shows my new numbers since changing to a 1K resistor from a 10K
I believe you may have mentioned that I should not use the array in the beginning of the sketch ?
The sensitivity is pretty important for the kick as it starts to play strangely without it (80)

``#include "MIDIcontroller.h"

// digitalelements FSR USB Module 8

byte MIDIchannel = 10;

const int pressPin [10] = {A0,A1,A2,A3,A4,A5,A6,A7,A8,A9}; // ANALOG pin

  // Pin & Note number
  MIDIdrum Pad0(A0, 36);
  MIDIdrum Pad1(A1, 38);
  MIDIdrum Pad2(A2, 69);
  MIDIdrum Pad3(A3, 65);
  MIDIdrum Pad4(A4, 69);
  MIDIdrum Pad5(A5, 67);
  MIDIdrum Pad6(A6, 65);
  MIDIdrum Pad7(A7, 96);
  MIDIdrum Pad8(A8, 101);
  MIDIdrum Pad9(A9, 75);

void setup(){

  // Input Range

  Pad0.inputRange(20, 380);
  Pad1.inputRange(20, 720);
  Pad2.inputRange(120, 860);
  Pad3.inputRange(120, 860);
  Pad4.inputRange(20, 720);
  Pad5.inputRange(20, 720);
  Pad6.inputRange(20, 720);
  Pad7.inputRange(20, 720);
  Pad8.inputRange(20, 720);
  Pad9.inputRange(20, 720);

  // Sensitivity

 //Pad0.sensitivity(80);
 // Pad1.sensitivity(99);
 // Pad2.sensitivity(99);
 // Pad3.sensitivity(99);
 // Pad4.sensitivity(99);
 // Pad5.sensitivity(99);
 // Pad6.sensitivity(99);
 // Pad7.sensitivity(99);
 // Pad8.sensitivity(99);
 // Pad9.sensitivity(99);

  // Velocity Output

  Pad0.outputRange(1,127);
  Pad1.outputRange(1,127);
  Pad2.outputRange(1,127);
  Pad3.outputRange(1,127);
  Pad4.outputRange(1,127);
  Pad5.outputRange(1,127);
  Pad6.outputRange(1,127);
  Pad7.outputRange(1,127);
  Pad8.outputRange(1,127);
  Pad9.outputRange(1,127);

  //WaitTime
  
  Pad0.setWaitTime(40);
  Pad1.setWaitTime(8);
  Pad2.setWaitTime(30);
  Pad3.setWaitTime(30);
  Pad4.setWaitTime(30);
  Pad5.setWaitTime(30);
  Pad6.setWaitTime(30);
  Pad7.setWaitTime(30);
  Pad8.setWaitTime(30);
  Pad9.setWaitTime(30);
  

}

void loop(){
  Pad0.send();
  Pad1.send();
  Pad2.send();
  Pad3.send();
  Pad4.send();
  Pad5.send();
  Pad6.send();
  Pad7.send();
  Pad8.send();
  Pad9.send();

// Crash Protection
  while(usbMIDI.read()){}

}
``

joshnishikawa, 2023-03-23:
> I got the warning below which looked like i was trying to access a parameter that was private ? ( sensitivity )

Sorry, I changed it to `setSensitivity()`. That jives better with `setWaitTime()`.

> I believe you may have mentioned that I should not use the array in the beginning of the sketch ?

I was saying you could delete `const int pressPin [10] = {A0,A1,A2,A3,A4,A5,A6,A7,A8,A9}; // ANALOG pin` since that never actually gets used in the sketch. Also `1, 127` is the default output range so you could omit those `Pad*.outputRange(1, 127)` calls. You could also set sensitivity in the constructor instead of using `setSensitivity().`

Try this.

```
#include "MIDIcontroller.h"

// digitalelements FSR USB Module 8

byte MIDIchannel = 10;

// Pin & Note number
MIDIdrum Pad0(A0, 36, 80);
MIDIdrum Pad1(A1, 38, 99);
MIDIdrum Pad2(A2, 69, 99);
MIDIdrum Pad3(A3, 65, 99);
MIDIdrum Pad4(A4, 69, 99);
MIDIdrum Pad5(A5, 67, 99);
MIDIdrum Pad6(A6, 65, 99);
MIDIdrum Pad7(A7, 96, 99);
MIDIdrum Pad8(A8, 101, 99);
MIDIdrum Pad9(A9, 75, 99);

void setup(){

// Input Range

Pad0.inputRange(20, 380);
Pad1.inputRange(20, 720);
Pad2.inputRange(120, 860);
Pad3.inputRange(120, 860);
Pad4.inputRange(20, 720);
Pad5.inputRange(20, 720);
Pad6.inputRange(20, 720);
Pad7.inputRange(20, 720);
Pad8.inputRange(20, 720);
Pad9.inputRange(20, 720);

//WaitTime

Pad0.setWaitTime(40);
Pad1.setWaitTime(8);
Pad2.setWaitTime(30);
Pad3.setWaitTime(30);
Pad4.setWaitTime(30);
Pad5.setWaitTime(30);
Pad6.setWaitTime(30);
Pad7.setWaitTime(30);
Pad8.setWaitTime(30);
Pad9.setWaitTime(30);

}

void loop(){
Pad0.send();
Pad1.send();
Pad2.send();
Pad3.send();
Pad4.send();
Pad5.send();
Pad6.send();
Pad7.send();
Pad8.send();
Pad9.send();

// Crash Protection
while(usbMIDI.read()){}

}
```

digitalelements, 2023-03-23:
Ahhh that's excellent !! .. thank you so much. I can see now that the sensitivity is built in with pin and note. sorry for note noticing that. you must be an early riser ? .. btw I did try to encapsulate my sketch with the proper characters to make it display properly ... has this changed again ?

joshnishikawa, 2023-03-23:
You have to use a triple backtick followed by a newline.
\`\`\`
```
code here
```
\`\`\`

digitalelements, 2023-03-23:
sorry :( got it now

digitalelements, 2023-03-24:
I've had a good go with the latest Library .. I did notice that the version number still showed 3.1.0 in this build in the arduino IDE. The code tweaks you've made with the multi drum sketch a very nice. I did notice I had to lower the Kick input range to start at 12. for some reason. it wants to be a bit lower to help eliminate the odd beater resting pressure on the sensor. The pedal start stop + pad sketch is outstanding. thank you kindly. I'll be doing more testing and tweaks over the weekend. learning to rename my Teensy so I can have all 3 devices connected at all times. btw did you happen o see the e-mail I sent about sensors ?

joshnishikawa, 2023-03-26:
This is weird. I'm sure I posted a reply to this but now it's not here.
> I did notice that the version number still showed 3.1.0

Thanks for that. If I don't get it right with the release, it'll be a nightmare.
> I did notice I had to lower the Kick input range to start at 12. for some reason. it wants to be a bit lower to help eliminate the odd beater resting pressure on the sensor.

Are you saying that you're getting extra triggers if the input range doesn't start low enough?

digitalelements, 2023-03-26:
It sure feels that way. I'll be spending more time with it to confirm. It oddly feels like it can trigger on the release of the FSR. I'm going to try and improve my plotter sketch so I can watch the input better. this sketch in the old IDE didn't jump around. not sure why it does in the new IDE.

joshnishikawa, 2024-06-15:
I think we've resolved the issue that this thread started on. Version 3.1.3 is officially released now so let's start a new thread for any problems that persist (or emerge) there.

closed 2024-06-15 by joshnishikawa