joshnishikawa/MIDIcontroller #32: Cannot run MIDIswitch_toggle

HaViGit, 2024-01-16:
Hi Josh,

First of all thanks for sharing this library, the MIDIpot example worked for me without a doubt.

Unfortunately I get error messages with the MIDIswitch_toggle example, below you can see the error code. I use Arduiono IDE 2.2.1 and a Teensy 4.0 board with board manager https://www.pjrc.com/teensy/package_teensy_index.json.

Can you help me solve this?

`c:/users/hans/appdata/local/arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Hans\AppData\Local\Temp\arduino\sketches\E4A21663C7284F73A7198036F2335F2C\libraries\Flicker\TouchSwitch.cpp.o: in function `TouchSwitch::setThreshold()':
c:\Users\Hans\Documents\Arduino\libraries\Flicker\src/TouchSwitch.cpp:29: undefined reference to `touchRead'
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "Bounce2.h"
  Used: C:\Users\Hans\Documents\Arduino\libraries\Bounce2
  Not used: C:\Users\Hans\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.58.1\libraries\Bounce2
exit status 1

Compilation error: exit status 1`

joshnishikawa, 2024-01-16:
Funny. Teensy 4.0 is supposed to ignore the bit of code that refers to touchRead because it doesn't feature capacitive touch. Can you post your sketch?

HaViGit, 2024-01-16:
Thanks for your quick response. The code is taken 1:1 from the MIDIswitch_toggle.ino demo (see below) of the library. Let us know if you need more information.

```
/*
  This example will show how to use a switch to toggle between two different
  MIDI messages. It could allow a switch to toggle START and STOP for example.
*/

#include "MIDIcontroller.h"

byte MIDIchannel = 5;
const int switchPin = 2; 
const int ledPin = 13; //Set an LED to show the state of the input.
bool state = false;

MIDIswitch myInput(switchPin, START); // Don't use LATCH

void setup(){
  pinMode(ledPin, OUTPUT);
}

void loop(){
  if ( myInput.send() > -1) { // no input == -1
    state = !state;
    if (state) myInput.setControlNumber(STOP);
    else myInput.setControlNumber(START);
  }

  digitalWrite(ledPin, state); // LED indicates state

// This prevents crashes that happen when incoming usbMIDI is ignored.
  while(usbMIDI.read()){}

// Also uncomment this if compiling for standard MIDI
//  while(MIDI.read()){}
}
```

joshnishikawa, 2024-01-16:
I see. It's probably confusing because I wrote START and STOP in all caps. Those aren't actually valid constants. You need to put two numbers between 1 and 127 there instead and then map the switch to the start and stop buttons in your DAW (if that's what you're trying to do).

joshnishikawa, 2024-01-16:
I forgot that I even implemented realtime messages. START and STOP should work fine and that wouldn't cause an error about touchRead anyway... Still thinking.

HaViGit, 2024-01-16:
A little more information about my wishes. The controller I am building is not that complex, for my software (Vengeance Avenger) I only want to use three sliders and two toggle buttons for the synth macros. The sliders now work with the sketch below, I would also like to use two toggle buttons.

```
#include "MIDIcontroller.h"

byte MIDIchannel = 5;

const int potPin1 = A0; 
const int potPin2 = A1;
const int potPin3 = A2;
const int ledPin  = 12;

MIDIpot myPot1(potPin1,  9);
MIDIpot myPot2(potPin2, 14);
MIDIpot myPot3(potPin3, 15);

void setup() {
  pinMode(ledPin, OUTPUT);
  digitalWrite(ledPin, LOW);
  Serial.begin(115200);
  delay(500);
  myPot1.outputRange(127, 0);
  myPot2.outputRange(127, 0);
  myPot3.outputRange(127, 0);
  digitalWrite(ledPin, HIGH);
}

void loop(){
  myPot1.send();
  myPot2.send();
  myPot3.send();
  while(usbMIDI.read()) {}
}
```

joshnishikawa, 2024-01-19:
Okay, I think I got it. The constructor that only takes a pin number and CC number was creating an unnecessary 'TouchSwitch' object. On boards that feature touchRead, the only problem is that it took up a bit of extra memory. Boards that don't feature touchRead threw errors. Should be fixed in version 3.1.2 now.

HaViGit, 2024-01-19:
Thanks Josh!

joshnishikawa, 2024-06-15:
The official release for version 3.1.3 of this library should more properly address any issues with using 4.x boards. Please open a new issue about any problems you encounter with that release.

closed 2024-06-15 by joshnishikawa