joshnishikawa/MIDIcontroller #33: MIDIswitch.ino example has error status 1

gtripper1970, 2024-06-11:
Hello Josh,
I recently installed the MIDIcontroller library and tried compiling the MIDIswitch.ino example code. I made no changes to the code. Unfortunately I've run into a strange error message:

[MIDIswitchErrorMessage.txt](https://github.com/user-attachments/files/15787015/MIDIswitchErrorMessage.txt)

Certainly appreciate any help you could provide.

Kind regards,
Gerry

gtripper1970, 2024-06-13:
UPDATE: The same error shows up regardless of the sketch content. Simply including the library in a blank sketch results in exactly the same error message.

joshnishikawa, 2024-06-13:
Thanks. Can you just post the error message as text so everyone can see what we're dealing with withoug having to download a file?

gtripper1970, 2024-06-13:
c:\Users\gwtri\Documents\Arduino\libraries\MIDIcontroller\src\MIDIdrum.cpp: In constructor 'MIDIdrum::MIDIdrum(int, byte, byte)':
c:\Users\gwtri\Documents\Arduino\libraries\MIDIcontroller\src\MIDIdrum.cpp:21:72: error: no matching function for call to 'TouchVelocity::TouchVelocity(int&, int, int)'
   21 | MIDIdrum::MIDIdrum(int p, byte num, byte type): TouchVelocity(p, 1, 127){
      |                                                                        ^
In file included from c:\Users\gwtri\Documents\Arduino\libraries\Flicker\src/Flicker.h:7,
                 from c:\Users\gwtri\Documents\Arduino\libraries\MIDIcontroller\src\MIDIdrum.h:5,
                 from c:\Users\gwtri\Documents\Arduino\libraries\MIDIcontroller\src\MIDIdrum.cpp:1:
c:\Users\gwtri\Documents\Arduino\libraries\Flicker\src/TouchVelocity.h:13:5: note: candidate: 'TouchVelocity::TouchVelocity(byte)'
   13 |     TouchVelocity(byte pin);
      |     ^~~~~~~~~~~~~
c:\Users\gwtri\Documents\Arduino\libraries\Flicker\src/TouchVelocity.h:13:5: note:   candidate expects 1 argument, 3 provided
c:\Users\gwtri\Documents\Arduino\libraries\Flicker\src/TouchVelocity.h:10:5: note: candidate: 'TouchVelocity::TouchVelocity()'
   10 |     TouchVelocity();
      |     ^~~~~~~~~~~~~
c:\Users\gwtri\Documents\Arduino\libraries\Flicker\src/TouchVelocity.h:10:5: note:   candidate expects 0 arguments, 3 provided
c:\Users\gwtri\Documents\Arduino\libraries\Flicker\src/TouchVelocity.h:6:7: note: candidate: 'TouchVelocity::TouchVelocity(const TouchVelocity&)'
    6 | class TouchVelocity{
      |       ^~~~~~~~~~~~~
c:\Users\gwtri\Documents\Arduino\libraries\Flicker\src/TouchVelocity.h:6:7: note:   candidate expects 1 argument, 3 provided
c:\Users\gwtri\Documents\Arduino\libraries\MIDIcontroller\src\MIDIdrum.cpp: In member function 'void MIDIdrum::outputRange(byte, byte)':
c:\Users\gwtri\Documents\Arduino\libraries\MIDIcontroller\src\MIDIdrum.cpp:123:20: error: 'setOutputRange' is not a member of 'TouchVelocity'
  123 |     TouchVelocity::setOutputRange(min, max);
      |                    ^~~~~~~~~~~~~~
c:\Users\gwtri\Documents\Arduino\libraries\MIDIcontroller\src\MIDIdrum.cpp: In member function 'void MIDIdrum::setThreshold()':
c:\Users\gwtri\Documents\Arduino\libraries\MIDIcontroller\src\MIDIdrum.cpp:142:20: error: 'setThreshold' is not a member of 'TouchVelocity'
  142 |     TouchVelocity::setThreshold();
      |                    ^~~~~~~~~~~~
Multiple libraries were found for "Encoder.h"
  Used: C:\Users\gwtri\Documents\Arduino\libraries\Encoder
  Not used: C:\Users\gwtri\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\libraries\Encoder
exit status 1

Compilation error: exit status 1

joshnishikawa, 2024-06-13:
Okay, I was afraid it was something like that. I'll work on fixing it this weekend. Can you tell me what board you're compiling for and what version of the library you're using?

gtripper1970, 2024-06-14:
Yes, of course. I have a Teensy 4.1 board and I'm using Version 2.5.4 of the library.

joshnishikawa, 2024-06-15:
Version 3.1.3 should be available in the Arduino library manager now. It's the first version that's compatible with Teensy 4.x . Give it a shot and let me know if you get any more errors.

gtripper1970, 2024-06-15:
Dear Josh,

I just tried Version 3.1.3 of the MIDIswitch.ino example. I get the following error:

C:\Users\gwtri\Downloads\MIDIcontroller-master\MIDIcontroller-master\examples\MIDIswitch\MIDIswitch.ino: In function 'void loop()':
C:\Users\gwtri\Downloads\MIDIcontroller-master\MIDIcontroller-master\examples\MIDIswitch\MIDIswitch.ino:53:9: error: 'MIDI' was not declared in this scope
   53 |   while(MIDI.read()){}
      |         ^~~~

exit status 1

Compilation error: 'MIDI' was not declared in this scope

A similar error pops up when I edit the last section of code to use "usbMIDI":

C:\Users\gwtri\Downloads\MIDIcontroller-master\MIDIcontroller-master\examples\MIDIswitch\MIDIswitch.ino: In function 'void loop()':
C:\Users\gwtri\Downloads\MIDIcontroller-master\MIDIcontroller-master\examples\MIDIswitch\MIDIswitch.ino:50:9: error: 'usbMIDI' was not declared in this scope
   50 |   while(usbMIDI.read()){}
      |         ^~~~~~~

exit status 1

Compilation error: 'usbMIDI' was not declared in this scope

Am I missing a library?

Cheers,
Gerry

joshnishikawa, 2024-06-15:
You probably just need to select something with MIDI from the Arduino tools->USB type menu.

gtripper1970, 2024-06-15:
Yep, for some reason the USB type reverted back to "Serial." Changed it to "MIDI," now the code compiles and installs properly. Successfully sent a MIDI CC message to my Bias FX software. It works!

Many thanks, Josh!

joshnishikawa, 2024-06-15:
Thanks for the feedback.

closed 2024-06-15 by joshnishikawa