joshnishikawa/MIDIcontroller #34: MIDIcontroller- Seeeduino Xiao SAMD21 support

digitalelements, 2024-06-18:
Hey Josh,

I'd like to make a request for adding Seeeduino Xiao to the list of compatible MCUs for your MIDIcontroller library.
A bonus may be that the Xiao SAMD21 supports touch pins as well.

[image]

Regards,

Chris

joshnishikawa, 2024-06-19:
It looks like a very nice piece of hardware. Is it able to be recognized by
a computer as a USB MIDI device and has someone written a USB MIDI library
for it? If so, the ‘wrapper’ library I wrote could be adapted but it would
be too tricky (at least for me) to accommodate multiple core libraries.
They each have different functions and handle them differently.

If you find out the seeeduino has a USB MIDI library, you should fork the
MIDIcontroller library and adapt it. It shouldn’t be that hard to do.

digitalelements, 2024-06-19:
Indeed it does get recognized as a USB MIDI device. I've taken things as far as adding some basic code to do an analog
read of a pin and send out a MIDI note on/off at a fixed velocity. it appears the basic Arduino MIDI Library as well as The Adafruit TinyUSB library are what most folks use to do USB MIDI with this series of board. I've observed the errors when trying to compile your MIDIdrum example to get clues as to what needs to be adjusted. Things like the "locked" to Teensy specific boards and features. elapsedMillis timer for instance. I'd love to attempt forking the library but may need a coach !
What's the best way to first break the Locked board warning ? .. I do see the board #define in the MIDIcontroller header file.

digitalelements, 2024-06-19:
This is the basic code that worked for me.

``` // Simple FSR analog read + MIDI Note output

#include <USB-MIDI.h>

USBMIDI_CREATE_DEFAULT_INSTANCE();

void setup() {

  Serial.begin(115200);
}

void loop() {

  // read the input on analog pin 0:
  int FSRinput = analogRead(A0);
  int Threshold = 20;

  // Observe the input:
  Serial.println(FSRinput);

  if (FSRinput >= Threshold){
  Serial.println ("Hit");
  }
    if (FSRinput > Threshold){
    MIDI.sendNoteOn(36, 64, 1);
    delay(10); 
    MIDI.sendNoteOff(36, 0, 1);
    }

  
  delay(1);        // delay in between reads for stability
}
```

joshnishikawa, 2024-06-21:
It looks like the SAMD21 chip covers lots of Arduino compatible boards that feature capacitive touch. So, updating Flicker and MIDIcontroller to support it would probably be worth doing. My XIAO should arrive tomorrow and I'll get to work.

digitalelements, 2024-06-21:
That's great news. I was hoping the touch Xiao features would be a bonus for your libraries. Thank you kindly ! And, I look forward to testing.