joshnishikawa/MIDIcontroller #35: Muxed-input and pot range settings question

NerdsWithAcid, 2024-11-28:
Hi Josh, How are you?
I've been (roughly still) building a midi controller to suite me a little bit better for live music shows, and building it myself suits me better than the one's comming from factory's. even the coding is more tailor made to my likings. 
Using the teensy-4.0 and did some testing all worked out all nicely. 
the thing that i didnt got to figure out yet, as i would like to make use of a mux-16 input, is the potentiometer and range minimum and maximun settings. 

This is possible in the example of the MIDIpot.ino, but iam struggeling on were to implement this in the useMuxedInput.ino
(if this is possible ?) i am quite struggling with this.

Thnx in advance, 
Greets.
Danny

joshnishikawa, 2025-01-10:
Sorry for the very late reply. I just can't seem to find any time to put into this project. What you're trying to do should be totally doable and straightforward. If you need different input and output ranges on specific pots, you just don't want to set them up in a loop. Set them up individually. Then just read them in a loop.

```
#include "MIDIcontroller.h"

byte MIDIchannel = 5;

const int selector_pin_a = 0;
const int selector_pin_b = 1;
const int selector_pin_c = 2;
const int selector_pin_d = 3; // uncomment only if using MUX-16

const int mux_input_pin = A0; // Change this to the ANALOG pin you want to use
MIDIpot* Pots[16]; // change to Pots[16] if using MUX-16

void setup(){
  pinMode(selector_pin_a, OUTPUT);
  pinMode(selector_pin_b, OUTPUT);
  pinMode(selector_pin_c, OUTPUT);
  pinMode(selector_pin_d, OUTPUT); // only if using MUX-16

  Pots[0] = new MIDIpot(mux_input_pin, 12);
  Pots[0]->inputRange(15, 1000); // helps reach the full output range
  Pots[0]->outputRange(30, 100); // maps the input range to the full MIDI range

  Pots[1] = new MIDIpot(mux_input_pin, 13);
  Pots[1]->inputRange(15, 1000);
  Pots[1]->outputRange(30, 100);

  Pots[2] = new MIDIpot(mux_input_pin, 14);
  Pots[2]->inputRange(15, 1000);
  Pots[2]->outputRange(30, 100);

  Pots[3] = new MIDIpot(mux_input_pin, 15);
  Pots[3]->inputRange(15, 1000);
  Pots[3]->outputRange(30, 100);

  Pots[4] = new MIDIpot(mux_input_pin, 16);
  Pots[4]->inputRange(15, 1000);
  Pots[4]->outputRange(30, 100);

  Pots[5] = new MIDIpot(mux_input_pin, 17);
  Pots[5]->inputRange(15, 1000);
  Pots[5]->outputRange(30, 100);

  Pots[6] = new MIDIpot(mux_input_pin, 18);
  Pots[6]->inputRange(15, 1000);
  Pots[6]->outputRange(30, 100);

  Pots[7] = new MIDIpot(mux_input_pin, 19);
  Pots[7]->inputRange(15, 1000);
  Pots[7]->outputRange(30, 100);

  Pots[8] = new MIDIpot(mux_input_pin, 20);
  Pots[8]->inputRange(15, 1000);
  Pots[8]->outputRange(30, 100);

  Pots[9] = new MIDIpot(mux_input_pin, 21);
  Pots[9]->inputRange(15, 1000);
  Pots[9]->outputRange(30, 100);

  Pots[10] = new MIDIpot(mux_input_pin, 22);
  Pots[10]->inputRange(15, 1000);
  Pots[10]->outputRange(30, 100);

  Pots[11] = new MIDIpot(mux_input_pin, 23);
  Pots[11]->inputRange(15, 1000);
  Pots[11]->outputRange(30, 100);

  Pots[12] = new MIDIpot(mux_input_pin, 24);
  Pots[12]->inputRange(15, 1000);
  Pots[12]->outputRange(30, 100);

  Pots[13] = new MIDIpot(mux_input_pin, 25);
  Pots[13]->inputRange(15, 1000);
  Pots[13]->outputRange(30, 100);

  Pots[14] = new MIDIpot(mux_input_pin, 26);
  Pots[14]->inputRange(15, 1000);
  Pots[14]->outputRange(30, 100);

  Pots[15] = new MIDIpot(mux_input_pin, 27);
  Pots[15]->inputRange(15, 1000);
  Pots[15]->outputRange(30, 100);
}

void loop(){
  for(int i=0; i<16; i++){ // change to i<16 if using MUX-16
    digitalWrite(selector_pin_d, (i&15)>>3); // uncomment only if using MUX-16
    digitalWrite(selector_pin_c, (i&7)>>2);
    digitalWrite(selector_pin_b, (i&3)>>1);
    digitalWrite(selector_pin_a, (i&1));

    Pots[i]->send();
  }  

// This prevents crashes that happen when incoming usbMIDI is ignored.
  while(usbMIDI.read()){}

// Also uncomment this if compiling for standard MIDI
//  while(MIDI.read()){}
}
```

Let me know if this helps you to get it working. If not, please let me know specifically what's going wrong or what error messages you're getting.

NerdsWithAcid, 2025-01-11:
Hi Josh, 
Thnx for this, i was looking for a startingpoint on this. 
I will give it a go, and post about it. 

Greets, 
Danny

closed 2026-08-21 by joshnishikawa